ros-tooling/aws-roadmap #248: Move `libstatistics_collector` package to separate repository

dabonnie, 2020-04-13:
## Description

Move the `libstatistics_collector' package into a separate repository. This is needed because 'libstatistics_collector' will be an rclcpp dependency.

## Test Plan

* Manual testing
  * Test the system_metrics_collector examples as it depends on this package
* Add `action-ros-ci` workflows for the project (copy the existing workflows)

## Documentation Plan

* README.md update for new repo
  * Add workflow badges
* Update system_metrics_collector readme with dependency info

## Release Plan

* create new repository with the package files

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

prajakta-gokhale, 2020-04-17:
Related PRs:
* [ ] https://github.com/ros-tooling/libstatistics_collector/pull/2: Add all collector interfaces from `system_metrics_collector` to new package `libstatictics_collector`
* [ ] Remove `libstatistics_collector` from `system_metrics_collector`

emersonknapp, 2020-04-24:
Reviewed at Sprint #17 review

closed 2020-04-24 by emersonknapp