ros-tooling/aws-roadmap #249: Move metrics_statistics_msgs package to separate repository

dabonnie, 2020-04-13:
## Description

Move the `metrics_statistics_msgs' package into a separate repository. This is needed because 'metrics_statistics_msgs' will be an rclcpp dependency.

## Test Plan

* Manual testing
  * Test the system_metrics_collector examples as it depends on this package

## Documentation Plan

* README.md update for new repo
  * note usage and how it relates to the C++ struct in 'libstatistics_collector'
* Update system_metrics_collector readme with dependency info

## Release Plan

* create new repository with the package files

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

prajakta-gokhale, 2020-04-17:
Related PRs:
* [ ] https://github.com/ros2/rcl_interfaces/pull/98: Add `metrics_statistics_msgs` messages that are in `system_metrics_collector` to `rcl_interfaces`
* [ ] https://github.com/ros-tooling/system_metrics_collector/pull/135: Remove `metrics_statistics_msgs` from `system_metrics_collector`

emersonknapp, 2020-04-24:
Moved to rcl_interfaces. Reviewed at Sprint #17 review

closed 2020-04-24 by emersonknapp