ros-tooling/aws-roadmap #250: Update ros2.repos file with new dependencies

dabonnie, 2020-04-14:
## Description

#225 requires new dependencies in ROS2 core. Before merging we should updated the core [ROS2 repos file](https://github.com/ros2/ros2/blob/master/ros2.repos). Note that this requires #248 and #249 to be merged.

## Test Plan

* Manual testing
  * checkout source via VCS and ensure everthing up to (at least) rclcpp can be built 

## Release Plan

* check in PR

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

prajakta-gokhale, 2020-04-24:
Related PRs: https://github.com/ros2/ros2/pull/897 and https://github.com/ros2/rcl_interfaces/pull/98

emersonknapp, 2020-04-24:
Reviewed at Sprint #17 review

closed 2020-04-24 by emersonknapp