ros-tooling/aws-roadmap #266: Investigate why pluginlib does not work on armhf

emersonknapp, 2020-05-07:
## Description

Since enabling the armhf builds last year - features that use class_loader/pluginlib do not work in that build, though they do build successfully.

This includes but is not limited to:
* rosbag2 plugins (and therefore its functionality)
* composable component nodes
* rviz and rqt plugins (not very important, it is not really expected that users run OpenGL GUIs on 32 bit platforms)

## Spike Timeboxing

2 days

## Spike Outcome

This spike is there to give time to explore the problem and identify the sources of the problem, if not their exact solutions.

 * Epic or stories to track the work of enabling this feature on armhf

## Goals

some notes
* it's probably something in Poco, or whatever we replaced it with for dynamic library loading.
* the build itself does not seem to work properly for rclcpp_components, in armhf builds the component binaries are not created, only the component libraries.

mm318, 2020-05-19:
Deprioritizing ARMHF-related issues.

emersonknapp, 2020-07-31:
Not currently supporting the armhf builds - closing this until it is on our roadmap

closed 2020-07-31 by emersonknapp