ros-tooling/aws-roadmap #272: ROS2 OSRF Testing

dabonnie, 2020-05-18:
## Description

Following the [ROS2 OSRF Testing Matrix](https://docs.google.com/spreadsheets/d/1EzDKMGoYopKuGWmzceXTVJ1zWqq1Wxw2jIGsh3CawAQ/edit#gid=1713002348), test the ROS2 tutorials with the following configuration:

 - system: OSX
 - RMW: FastRTPS, Connext, CycloneDDS

Progress for each item should be tracked on the OSRF Testing Matrix per their guidelines.

## Spike Timeboxing

3 days

## Spike Outcome

 * Best effort testing of the ROS2 tutorials using OSX.

thomas-moulard, 2020-06-05:
There are a couple comments for those tasks, but they are marked as successful. Did we report those issues to OR?

```
f8ffc2502b19:ros2-osx snehaldb$ ros2 topic echo /turtle1/color_sensor
WARNING: topic [/turtle1/color_sensor] does not appear to be published yet
1590601601.311893 [0] 359871: using network interface en0 (udp/192.168.0.14) selected arbitrarily from: en0, utun1
Traceback (most recent call last):
File "/Users/snehaldb/ros2_foxy/ros2-osx/bin/ros2", line 11, in <module>
load_entry_point('ros2cli==0.9.3', 'console_scripts', 'ros2')()
File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
rc = extension.main(parser=parser, args=args)
File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/ros2topic/command/topic.py", line 41, in main
return extension.main(args=args)
File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/ros2topic/verb/echo.py", line 81, in main
return main(args)
File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/ros2topic/verb/echo.py", line 98, in main
subscriber(
File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/ros2topic/verb/echo.py", line 110, in subscriber
node.create_subscription(
File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/rclpy/node.py", line 1201, in create_subscription
check_for_type_support(msg_type)
File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/rclpy/type_support.py", line 20, in check_for_type_support
ts = msg_type.__class__._TYPE_SUPPORT
AttributeError: type object 'NoneType' has no attribute '_TYPE_SUPPORT' This might be a ROS 1 message type but it should be a ROS 2 message type. Make sure to source your ROS 2 workspace after your ROS 1 workspace.
```

thomas-moulard, 2020-06-05:
Could you confirm all tutorials were OK?

zmichaels11, 2020-06-05:
missing cyclone?

zmichaels11, 2020-06-05:
Were tickets opened for an issues found?

emersonknapp, 2020-06-15:
Pinging this for progress, hasn't moved in a while

closed 2020-06-18 by zmichaels11