ros-tooling/aws-roadmap #274: ROS2 Foxy QA

dabonnie, 2020-05-20:
## Description

Support the ROS2 Foxy release by providing QA for the core packages and AWS Foundations / ros-tooling owned features. These features include:

 - rclcpp topic statistics
 - rosbag2 splitting
 - rosbag2 QoS
 - incompatible QoS callbacks

emersonknapp, 2020-06-01:
High level callout for OSX setup instructions: https://github.com/ros2/ros2_documentation/issues/727 the RTI connext link is incorrect

emersonknapp, 2020-06-01:
High level callout for Windows setup instructions: Python version is wrong in instructions https://github.com/ros2/ros2_documentation/pull/723

emersonknapp, 2020-06-02:
More windows setup instructions fixes https://github.com/ros2/ros2_documentation/pull/732

emersonknapp, 2020-07-31:
Done!

closed 2020-07-31 by emersonknapp