ros-tooling/aws-roadmap #275: ROS2 Foxy Rosbag Splitting QA

dabonnie, 2020-05-20:
## Description

Record high volume data using rosbag 2 record with splitting enabled for various sizes. Verify that the split files can be played back.

The following configurations should be tested to ensure the feature works on all platforms and middlewares:

|  *Operating System* |
|-------------------|
|   OSX |
|  Windows |
|  Linux x86_64 |  
|  Linux aarch64 |  

|  *RMW* |
|-------------------|
|   FastRTPS |
|  Connext |
|  CycloneDDS |  

## Spike Timeboxing

2 days

## Spike Outcome

 * github tickets if any bugs are found
 * document (or comment update here) how to reproduce manual tests

piraka9011, 2020-06-02:
To reproduce:

Install the prerequisites and [fat binaries](https://index.ros.org/doc/ros2/Installation/Prerelease-Testing/) on the target platform. 

```
ros2 topic pub /test_topic std_msgs/String "data: Hello World"
# TODO
```

- Linux
  - FastRTPS
    - [ ] record
    - [ ] playback
  - RTI Connext
    - [ ] record
    - [ ] playback
  - Cyclone DDS
    - [ ] record
    - [ ] playback
- OSX
  - FastRTPS
    - [ ] record
    - [ ] playback
  - RTI Connext
    - [ ] record
    - [ ] playback
  - Cyclone DDS
    - [ ] record
    - [ ] playback
- Windows
  - FastRTPS
    - [ ] record
    - [ ] playback
  - RTI Connext
    - [ ] record
    - [ ] playback
  - Cyclone DDS
    - [ ] record
    - [ ] playback

piraka9011, 2020-06-03:
Opened ros2/rosbag2#428 to enhance the UX when specifying a split size.

prajakta-gokhale, 2020-06-05:
Opened https://github.com/ros2/rosbag2/issues/432 for incorrect behaviour on setting invalid splitting size, when using splitting and compression together.

closed 2020-06-18 by zmichaels11