ros-tooling/aws-roadmap #277: ROS2 Foxy Incompatible QoS QA: Test Policies

dabonnie, 2020-05-20:
## Description

Use publishers and subscriptions with compatible and incompatible QoS policies.

The following configurations should be tested to ensure the feature works on all platforms and middlewares:

|  *Operating System* |
|-------------------|
|   OSX |
|  Windows |
|  Linux x86_64 |  
|  Linux aarch64 |  

|  *RMW* |
|-------------------|
|   FastRTPS |
|  Connext |
|  CycloneDDS |  

## Spike Timeboxing

1 days

## Spike Outcome

 * github tickets if any bugs are found
 * document (or comment update here) how to reproduce manual tests

emersonknapp, 2020-05-21:
Is this ticket supposed to be about the QoS Incompatibility _callbacks_?

dabonnie, 2020-05-21:
No need to iterate through all policies, but try a few compatible and incompatible policies.

dabonnie, 2020-05-28:
- Linux
  - FastRTPS
    - [ ] rclcpp
    - [ ] rclpy
  - RTI Connext
    - [ ] rclcpp
    - [ ] rclpy
  - Cyclone DDS
    - [ ] rclcpp
    - [ ] rclpy
- OSX
  - FastRTPS
    - [ ] rclcpp
    - [ ] rclpy
  - RTI Connext
    - [ ] rclcpp
    - [ ] rclpy
  - Cyclone DDS
    - [ ] rclcpp
    - [ ] rclpy
- Windows
  - FastRTPS
    - [ ] rclcpp
    - [ ] rclpy
  - RTI Connext
    - [ ] rclcpp
    - [ ] rclpy
  - Cyclone DDS
    - [ ] rclcpp
    - [ ] rclpy

To reproduce:

Install the perquisites and [fat binaries](https://index.ros.org/doc/ros2/Installation/Prerelease-Testing/) on the desired system. 

`rclcpp`
 - compatible: `ros2 run quality_of_service_demo_cpp liveliness 1000 --kill-publisher-after 2000`
 - incompatible: `ros2 run quality_of_service_demo_cpp incompatible_qos durability`

closed 2020-06-05 by zmichaels11