ros-tooling/aws-roadmap #279: QA Topic Statistics Configuration

dabonnie, 2020-05-20:
## Description

Verify options (topic name, publish period, enable / disable state) are configurable.

The following configurations should be tested to ensure the feature works on all platforms and middlewares:

|  *Operating System* |
|-------------------|
|   OSX |
|  Windows |
|  Linux x86_64 |  
|  Linux aarch64 |  

|  *RMW* |
|-------------------|
|   FastRTPS |
|  Connext |
|  CycloneDDS |  

## Spike Timeboxing

0.5 days

## Spike Outcome

 * github tickets if any bugs are found
 * document (or comment update here) how to reproduce manual tests

dabonnie, 2020-05-21:
Take code like the talker demo and manually change the subscription options.

emersonknapp, 2020-05-28:
- Linux
  - FastRTPS
    - [x] rclcpp
    - [x] rclpy
  - RTI Connext
    - [x] rclcpp
    - [x] rclpy
  - Cyclone DDS
    - [x] rclcpp
    - [x] rclpy
- OSX
  - FastRTPS
    - [ ] rclcpp
    - [ ] rclpy
  - RTI Connext
    - [ ] rclcpp
    - [ ] rclpy
  - Cyclone DDS
    - [ ] rclcpp
    - [ ] rclpy
- Windows
  - FastRTPS
    - [ ] rclcpp
    - [ ] rclpy
  - RTI Connext
    - [ ] rclcpp
    - [ ] rclpy
  - Cyclone DDS
    - [ ] rclcpp
    - [ ] rclpy

To reproduce:

Install the prerequisites and [fat binaries](https://index.ros.org/doc/ros2/Installation/Prerelease-Testing/) on the target platform.

dabonnie, 2020-05-29:
FYI we only implemented in rclcpp and not rclpy.

emersonknapp, 2020-05-30:
Progress tracking here because markdown is hard https://docs.google.com/spreadsheets/d/1xlDpkV_n9cCc9rOpJBsJ5KNkfjGQDXItqp4XY18oyOI/edit#gid=1713002348

emersonknapp, 2020-05-30:
Testing notes - Linux:

* Build retrieved from:
  * ci.ros2.org
* Build test workspace
  * `source ros2-linux/setup.bash`
  * Clone this repository https://github.com/emersonknapp/foxytest into src/
  * `colcon build`
* Run test, each command in separate shell
  * `ros2 topic pub /chatter std_msgs/String "data: hello"`
  * `ros2 run topic_statistics_test chatter_stats`
  * `ros2 topic echo ${STATS_TOPIC}`

Twiddle values in chatter_stats.cpp

emersonknapp, 2020-05-30:
Enable/disable: 
* set on, see topic pub
* set off, see no topic

Publish Period:
Expectation: `ros2 topic hz /statistics` gives `2 * 1 / period` (2 messages are published each cycle)
* Works: 2s, 500ms, 1ms
* Fails at compile: 20us
* Fails at runtime: -1s
* 0ms publishes at apparently maximum loop speed

Topic Name
Expectation: `ros2 topic echo topicname` prints messages
* Works: "/staty", "tostastics", "/topic/stats"
* Fails at runtime: "", "top-ic"

emersonknapp, 2020-05-30:
Mac fat binary: https://ci.ros2.org/view/packaging/job/packaging_osx/1865/artifact/ws/ros2-package-osx-x86_64.tar.bz2

Follow setup instructions at https://index.ros.org/doc/ros2/Installation/Foxy/macOS-Install-Binary/ (skip rviz rqt sros deps)
No other differences from Linux

emersonknapp, 2020-05-30:
Windows: using https://ci.ros2.org/view/packaging/job/packaging_windows/1754/artifact/ws/ros2-package-windows-AMD64.zip

Followed setup instructions for https://index.ros.org/doc/ros2/Installation/Foxy/Windows-Install-Binary/ - skipping opencv and qt stuff

Using "Developer Command Prompt for VS2019" for all commands - multiple windows open simultaneously sourcing the same build. Building same repo as other platforms after running `.\ros2-windows\setup.bat`

emersonknapp, 2020-05-30:
Created https://github.com/ros2/ros2_documentation/pull/723 to enable windows testing (using manual packaging build from https://ci.ros2.org/view/packaging/job/packaging_windows/1754/)

dabonnie, 2020-06-05:
Patch as a result of this testing: https://github.com/ros2/rclcpp/pull/1151#pullrequestreview-424032892

closed 2020-06-05 by zmichaels11