ros-tooling/aws-roadmap #280: QA Topic Statistics No Timestamp

dabonnie, 2020-05-20:
## Description

Obtain statistics for various messages that do not use a timestamped message header (majority of ROS messages, e.g. run the talker / subscriber demo and enable statistics).

The following configurations should be tested to ensure the feature works on all platforms and middlewares:

|  *Operating System* |
|-------------------|
|   OSX |
|  Windows |
|  Linux x86_64 |  
|  Linux aarch64 |  

|  *RMW* |
|-------------------|
|   FastRTPS |
|  Connext |
|  CycloneDDS |  

## Spike Timeboxing

0.5 days

## Spike Outcome

 * github tickets if any bugs are found
 * document (or comment update here) how to reproduce manual tests

emersonknapp, 2020-05-30:
Setup

* Build retrieved from:
  * ci.ros2.org
* Build test workspace
  * `source ros2-linux/setup.bash`
  * Clone this repository https://github.com/emersonknapp/foxytest into src/
  * `colcon build`
* Run test, each command in separate shell
  * `ros2 topic pub /chatter std_msgs/String "data: hello"`
  * `ros2 run topic_statistics_test chatter_stats`
  * `ros2 topic echo ${STATS_TOPIC}`

Twiddle values in chatter_stats.cpp

emersonknapp, 2020-05-30:
Opened 
https://github.com/ros-tooling/libstatistics_collector/issues/28
https://github.com/ros-tooling/libstatistics_collector/issues/27

emersonknapp, 2020-05-30:
Opened https://github.com/ros-tooling/libstatistics_collector/issues/29

emersonknapp, 2020-05-30:
Other than behavioral uncertainty no issues, seems to be working. Used same setup as per #279

emersonknapp, 2020-06-04:
Was able to test Connext on Windows by using the eval license from the Linux install - all working as expected

closed 2020-06-05 by zmichaels11