ros-tooling/aws-roadmap #282: QA Rosbag 2 QoS: Mixed Policies

dabonnie, 2020-05-21:
## Description

Record and playback a [mixed](https://github.com/ros-tooling/aws-roadmap/issues/217) QoS topic

The following configurations should be tested to ensure the feature works on all platforms and middlewares:

|  *Operating System* |
|-------------------|
|   OSX |
|  Windows |
|  Linux x86_64 |  
|  Linux aarch64 |  

|  *RMW* |
|-------------------|
|   FastRTPS |
|  Connext |
|  CycloneDDS |  

## Spike Timeboxing

1 days

## Spike Outcome

 * github tickets if any bugs are found
 * document (or comment update here) how to reproduce manual tests

prajakta-gokhale, 2020-06-03:
## Installation
* [Fat binaries](https://ci.ros2.org/view/packaging/) for respective platforms
* [RTI Connext DDS](https://index.ros.org/doc/ros2/Installation/DDS-Implementations/) installation for respective platforms
* [Dependencies](https://index.ros.org/doc/ros2/Installation/Foxy/#binary-packages) for respective platforms

## Test setup
1. Create 2 publishers with different QoS settings publishing to the same topic
```bash
> ros2 topic pub /talker --qos-durability volatile std_msgs/String "data: Hello World volatile"
> ros2 topic pub /talker --qos-durability transient_local std_msgs/String "data: Hello World transient_local"
```
2. Verify the 2 publishers, validate the QoS part
```bash
> ros2 topic info /talker -v
```
3. Record with default rosbag2 QoS. A `WARN` log should appear stating which QoS policy is being used to record.
```bash
> ros2 bag record -a -o $filename
...
[WARN] [1591201849.252359669] [rosbag2_transport]: Some, but not all, publishers on topic "/talker" are offering RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL. Falling back to RMW_QOS_POLICY_DURABILITY_VOLATILE as it will connect to all publishers. Previously-published latched messages will not be retrieved.
```
4. Verify that messages from both publishers were recorded, validate the `offered_qos_profiles` part
```bash
> less $filename/metadata.yaml
```
5. Play the recorded bagfile
```bash
> ros2 bag play $filename
```
6. Monitor the playback topic, validate that messages from both publishers are played back. A `WARN` log should appear stating which QoS policy is being used to replay.
```
> ros2 topic echo /talker
...
[WARN] [1591201959.178510698] [rosbag2_transport]: Not all original publishers on topic /talker offered the same QoS profiles. Rosbag2 cannot yet choose an adapted profile to offer for this mixed case. Falling back to the rosbag2_transport default publisher offer.
```

prajakta-gokhale, 2020-06-03:
## Test results

1. Linux

| Feature  | FastRTPS | CycloneDDS | RTI Connext DDS |
| ------------- | ------------- | ------------- | ------------- |
| Record  | ✅  | ✅   | ✅  |
| Play  | ✅  | ✅  | ✅  |

2. macOS

| Feature  | FastRTPS | CycloneDDS | RTI Connext DDS |
| ------------- | ------------- | ------------- | ------------- |
| Record  |  |  |  |
| Play  |  |  |  |

3. WIndows

| Feature  | FastRTPS | CycloneDDS | RTI Connext DDS |
| ------------- | ------------- | ------------- | ------------- |
| Record  |  |  |  |
| Play  |  |  |  |

closed 2020-06-18 by zmichaels11