ros-tooling/aws-roadmap #285: QA Rosbag 2 QoS: Non-default QoS Profiles

dabonnie, 2020-05-21:
## Description

Record and playback data with a non-default QoS setting, e.g., ```RELIABILITY_BEST_EFFORT```.

The following configurations should be tested to ensure the feature works on all platforms and middlewares:

|  *Operating System* |
|-------------------|
|   OSX |
|  Windows |
|  Linux x86_64 |  
|  Linux aarch64 |  

|  *RMW* |
|-------------------|
|   FastRTPS |
|  Connext |
|  CycloneDDS |  

## Spike Timeboxing

1 days

## Spike Outcome

 * github tickets if any bugs are found
 * document (or comment update here) how to reproduce manual tests

jikawa-az, 2020-05-29:
To reproduce:

Install the perquisites and [fat binaries](https://index.ros.org/doc/ros2/Installation/Prerelease-Testing/) on the target platform. 

`rclcpp`
 - Source ROS2 in a new terminal and run `ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 2000`
 - Source ROS2 in a new terminal and run `ros2 bag record /qos_deadline_chatter`
 - Close both with ctrl+c
 - Run `ros2 bag play <bag-file-name>` in one window and run `ros2 topic echo /qos_deadline_chatter` immediately in another window
'rclpy'
 - `ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 2000`

To use [Cyclone DDS](https://github.com/ros2/rmw_cyclonedds): `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`

To use [RTI Connext](https://index.ros.org/doc/ros2/Installation/DDS-Implementations/): `cd /opt/rti.com/rti_connext_dds-5.3.1/resource/scripts && source ./rtisetenv_x64Linux3gcc5.4.0.bash; cd -
`
- Linux
  - FastRTPS
    - [] rclcpp
    - [] rclpy
  - RTI Connext
    - [] rclcpp
    - [] rclpy
  - Cyclone DDS
    - [] rclcpp
    - [] rclpy
- OSX
  - FastRTPS
    - [] rclcpp
    - [ ] rclpy
  - RTI Connext
    - [ ] rclcpp
    - [ ] rclpy
  - Cyclone DDS
    - [] rclcpp
    - [] rclpy
- Windows
  - FastRTPS
    - [ ] rclcpp
    - [ ] rclpy
  - RTI Connext
    - [ ] rclcpp
    - [ ] rclpy
  - Cyclone DDS
    - [ ] rclcpp
    - [ ] rclpy

dabonnie, 2020-06-03:
Verified RTI Connext on OSX.

dabonnie, 2020-06-04:
Verified on Windows.

jikawa-az, 2020-06-04:
Verified on Linux

closed 2020-06-05 by zmichaels11