ros-tooling/aws-roadmap #291: QA CLI Node Info

dabonnie, 2020-05-22:
## Description

Use command line tools and launch files to verify duplicate node names and enhanced QoS info per:

 - https://github.com/ros2/ros2cli/pull/463
 - https://github.com/ros2/ros2cli/pull/462
 - https://github.com/ros2/launch_ros/pull/127
 - https://github.com/ros2/ros2/issues/830

The following configurations should be tested to ensure the feature works on all platforms and middlewares:

|  *Operating System* |
|-------------------|
|   OSX |
|  Windows |
|  Linux x86_64 |  

|  *RMW* |
|-------------------|
|   FastRTPS |
|  Connext |
|  CycloneDDS |  

**Priority**
 - Linux + default RMW, non default RMW
 - Windows + default RMW, non default RMW
 - OSX + default RMW, non default RMW

## Spike Timeboxing

0.5 days

## Spike Outcome

 * github tickets if any bugs are found
 * document (or comment update here) how to reproduce manual tests

prajakta-gokhale, 2020-05-27:
With default rmw implementation (fastrtps):

| Feature  | Linux | macOS | Windows |
| ------------- | ------------- | ------------- | ------------- |
| `ros2 node list`  | ✅   | ✅  |. |
| `ros2 node info`  | ✅   | ✅  |. |
| `launch` log WARNING |. |. |. |

prajakta-gokhale, 2020-06-01:
Using fat binary generated by https://ci.ros2.org/view/packaging/job/packaging_windows/1758/

closed 2020-06-18 by zmichaels11